adjacentlink/emane #190: Add RHEL 8 build target

dpward, 2021-03-03:
Hi all,

Can we please add RHEL 8 as a build target? It was released on May 2019 and has already had 3 minor releases.

(Just an aside: while RHEL 7 will continue to be supported until June 2024, it has reached the end of Full Support as well as Maintenance Support 1. RHEL 7.9 was the final minor release. It no longer receives updates for new hardware enablement, which by itself might motivate some organizations to transition. I'm not suggesting this be dropped as a build target, but just providing this FYI.)

Thanks!

sgalgano, 2021-03-03:
Are you encountering an issue building rpms for RHEL8. Our CentOS 8 build environment is not reporting any issues? The current spec file supports el8.

dpward, 2021-04-14:
Sorry for the delay. I was referring to the official builds that can be downloaded from the website. They seem to be missing the CentOS 8 builds.

bharnden, 2021-04-14:
@dpward I would assume with the switch to stop supporting CentOS 8 early, it would probably not be worth supporting.

dpward, 2021-04-14:
@bharnden I was actually asking about using this on RHEL 8, which will continue to be supported until 2029. Also keep in mind:
https://www.redhat.com/en/blog/introducing-red-hat-universal-base-image

RHEL 7 has already reached the point where they are not making changes to it, and there is pressure to migrate away from it.

b00ga, 2021-04-14:
So a couple of things to @dpward 's point:
* Red Hat Enterprise Linux is really the thing being requested for targeting, and is fully supported by Red Hat. The latest major version, 8, will be supported until 2029. See https://access.redhat.com/support/policy/updates/errata#RHEL8_Life_Cycle
* There are a number of projects that rebuild and/or customize RHEL, including current version 8. One commercial option in Oracle Linux 8. One extremely popular free version is CentOS 8. Yes, C8 goes end-of-support Dec 2021. CentOS 8 (a "bug for bug" rebuild of RHEL8) is being replaced with CentOS Stream 8. Stream will be supported for basically the lifecycle of RHEL8, and is now the UPSTREAM distribution to RHEL 8 point releases. There are also a number of competing rebuilds looking to take the place of the "bug for bug" rebuild that CentOS 8 _was_ and likely one will shake out as a CentOS replacement, including but not limited to Alma Linux (https://almalinux.org/), Rocky Linux (https://rockylinux.org/),  and Navy Linux (https://navylinux.org/). So RHEL8 should really still be supported and that provides packages for all these RHEL-derived downstream distros as well.
* Free licenses for RHEL are available for development. I addition to the UBI that David mentions (a container baseline RHEL8 image), organization can apply for free licenses for development. See https://developers.redhat.com/articles/faqs-no-cost-red-hat-enterprise-linux

sgalgano, 2021-12-30:
Release bundles for [Rocky Linux 8](https://github.com/adjacentlink/emane/wiki/Install#rhelrocky-8-x86_64) and [CentOS Stream 8](https://github.com/adjacentlink/emane/wiki/Install#centos-stream-8-x86_64) are now available.

closed 2021-12-30 by sgalgano