adjacentlink/emane #191: [Question] How to retrieve pathloss and/or SINR between two nodes?

lemon-lyman, 2021-03-05:
Is there a way to retrieve metrics like pathloss and SINR between two specific nodes at any given time in an emulation?

We need to programmatically publish these metrics to an external source at a high frequency during a running emulation. We looked at calling emanesh and scraping the output from the command line, but emanesh only seems to provide averages of some of these values (like SINR) and we need the instantaneous SINR. Ideally this would be a builtin python function.

Using EMANE 1.2.5 and CORE 7.2.1 both inside a ubuntu docker container

sgalgano, 2021-03-05:
You can communicate directly with a running emulator instance using the [emane control port](https://github.com/adjacentlink/emane-control-port-tutorial). This is what `emanesh` uses under the hood.

For SINR, we typically calculate the average SINR for a given transmitter at a receiver over the query window, using the long running average provided in the NeighborMetricTable. Where, the window is the delta of two successive queries, using the current (cur) and previous (prev) query values: 

```
SINR_window = ((Total_Pkts_Rx_cur * SINR_Mean_cur) - (Total_Pkts_Rx_prev * SINR_Mean_prev)) / 
                           (Total_Pkts_Rx_cur - Total_Pkts_Rx_prev)
```

Remembering to be mindful of when the window receive packet count is 0.

Only `precomputed` pathloss is available via a table in the 1.2.x physical layer. The current develop branch contains [additions]( https://github.com/adjacentlink/emane/commit/0c183c4549993482d58c850d8b87cc39666c74fd) to the ReceivePowerTable which include pathloss calculated when  `propagationmodel` is `freespace` or `2ray`.  Note: the ReceivePowerTable shows values for the last received over-the-air message.

lemon-lyman, 2021-03-05:
Thanks for the help! I will give this a try.

I hadn't seen that emane control port tutorial before. Is it linked somwhere in the repo and I just missed it?

closed 2021-03-16 by lemon-lyman