adjacentlink/emane #193: [Question] Possible to change antenna attitude at runtime?

lemon-lyman, 2021-03-16:
Is it possible to update the antenna attitude/orientation at runtime? The information isn't know beforehand so an EEL file doesn't seem like an option.

sgalgano, 2021-03-16:
Yes, send an [AntennaProfileEvent](https://github.com/adjacentlink/emane/blob/v1.2.7/src/python/emane/events/antennaprofileevent.py) from an external application:

```python
from emane.events import EventService
from emane.events import AntennaProfileEvent

# create the event service
service = EventService(('224.1.2.8',45703,'lo'))

# create an antenna profile event to point nem 1 and send to all nems
event = AntennaProfileEvent()
event.append(1,profile=1,azimuth=105.0,elevation=10.0)

# publish the event
service.publish(0,event)
```

You can verify the above example using `emaneevent-dump`:

```
$ emaneevent-dump -i lo
[1615908331.558573] nem: 0 event: 102 len: 24 seq: 1 [AntennaProfile]
 UUID: fb575e7c-af88-4c96-b6ce-4c017ef45ff9
   (1, {'profile': 1, 'azimuth': 105.0, 'elevation': 10.0})
```

lemon-lyman, 2021-03-16:
Thanks for the response!

Should I expect these changes to show in the LocationEventInfoTable? After appending your code to my existing CORE script, the LocationEventInfoTable seems to stay as the default values.

[image]

I also tried publishing a LocationEvent and it also had no effect on the LocationEventInfoTable

I'm using CORE 7.1, EMANE 1.2.5 in Ubuntu

sgalgano, 2021-03-16:
Try AntennaProfileEventInfoTable.

lemon-lyman, 2021-03-16:
AntennaProfileEventInfoTable looks empty

[image]

sgalgano, 2021-03-16:
Did you adjust the script to match the [interface](https://github.com/adjacentlink/emane/wiki/Configuring-the-Emulator#eventservicedevice) the emulator is listening for events on?

lemon-lyman, 2021-03-17:
It appears to be the same as in the code provided. Here are sections of the core-daemon output

[image]

and the emane log on the node
[image]

lemon-lyman, 2021-03-17:
I realized my mistakes.

1. I did not change the device to 'ctrl0' from the 'lo' used in your example
2. I was running the script on host instead of inside the 'nodes' (namespaces) created by CORE

I fixed these two problems and it works now, thanks.

closed 2021-03-17 by lemon-lyman