adjacentlink/emane #194: Questions about IP header TOS value changes in EMANE

szl0144, 2021-04-16:
Hi,

OS: ubuntu18.04
mgen: 5.02b
EMANE:v1.2.6
EMANE tutorial:v0.2.8 

I want to use two reserved bits in the 8 bits Type of service (TOS) field of IP header, I change the TOS value from default value 0 to 1 using mgen. But I found emanenode0 doesn't transmit any packets to nodes container, there are only some TCP, ARP and HTTPS handshakes between emanenode0 and nodes without any connections. Can you please whether there is any rule blocking packets whose TOS is not 0 in EMANE? 

Thank you very much!

Best,
Shaoyi

sgalgano, 2021-04-20:
There is no logic in the emulator or the open source radio models to drop packets based on TOS.

szl0144, 2021-05-06:
Thanks, I will go to parse the packets from MGEN to see where is the problem.

closed 2021-05-06 by szl0144

szl0144, 2021-05-07:
After the troubleshoot, I can receive the packets whose TOS value is 1 in EMANE container right now. Thanks!