adjacentlink/emane #195: EMANE topology changes using democtl-host script

szl0144, 2021-04-16:
Hi,

I am trying to create a multi-hop topology where the traffics goes from node-1 to node-2, then goes from node-2 to node-3. Can I use the democrats-host script in the demo-start file to let the emanenode0 bridge only connects with node 1 (pass the node number parameter 1 into the script). Then adding a static route from node-1 to node-2 in the routing table of node 1 as well as adding a static route from node-2 to node-3 in the routing table of node 3? I am not sure if node-1 and node-2 will forward packets if I just connect one node to the emanenode0. Can you please give me some suggestions? Thanks!

Best,
Shaoyi

sgalgano, 2021-04-20:
You can use pathloss events or locations events (when using [freespace or 2ray](https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#propagationmodel)) to change topology. You can use static routes or you can use a MANET routing protocol. The [emane-tutorial](https://github.com/adjacentlink/emane-tutorial) uses olsr. The [letce2-tutorial](https://github.com/adjacentlink/letce2-tutorial) uses batman-adv by default.

szl0144, 2021-05-06:
> You can use pathloss events or locations events (when using [freespace or 2ray](https://github.com/adjacentlink/emane/wiki/Physical-Layer-Model#propagationmodel)) to change topology. You can use [...]

szl0144, 2021-05-06:
Hi,

Thank you for your suggestions. I am using static routes method to change the topology, I notice the traffic source in MGEN is a port without an IP address. Because I am going to set up a traffic from node-1 to node-3, all three nodes send packets if I set a port as SRC. can you please tell me how to set a source IP in EMANE?  I can set the DST address (10.99.0.3/5001) for node-3 in MGEN right now.

I also notice the olsr link viewer shows the links between nodes (10.100.0.X), is the link the same as the routing traffic between radios? Because I see the node's address is different from the radio's address and I need to set up the radio's address in the routing table referring to https://github.com/adjacentlink/emane-tutorial/wiki#lxc-test-node-topology, I am not sure whether the viewer shows the routing traffic.  

Thank you very much.
Best,