adjacentlink/emane #202: Feature: add ability to pause, resume and jump to different time in event generator

andreas-ibm, 2021-07-27:
So I'm running in a long scenario, and see something weird with the routing protocol around the 1hr mark, so I'd like to jump to  5 minutes before and replay to see if it happens again.  Currently I have to kill the run, chop my .eel file(s) down to that segment, re-number the .eel file, and then re-run.  If I could send the daemon a signal (via network or otherwise) to go to a particular time offset then that would be great.

Also... In this long scenario, I need a "setup" phase, after sending the events for time offset 0 (to establish location and pathloss between the nodes so they can talk to each other) I might need to start routing and let it stabilise, or initialise a bunch of containers on the network.  So it would be cool if I could pause the events after a particular timestep and then resume when everything is ready for my actual experiment!

I was about to reimplement an .eel loader / generator in Node.js when I thought I should probably look at whether it's easier to modify emaneeventservice to do what I want... Do you think it's conceivable, or would it require too much re-engineering of the code?

sgalgano, 2021-07-27:
You might want to take a look at using the `emane.events` Python module to build a custom controllable event distribution application for your needs. [Here](https://github.com/adjacentlink/emane-tutorial/blob/master/1/bridge_with10.py) is a bare bones example for sending an event.

andreas-ibm, 2021-07-27:
yeah, that's the direction I was going in (though using node.js spitting out protobuf events), but the .eel files are massive (up to one and a half million lines of location and pathloss events) so it would be annoying to have to effectively reimplement all the other stuff in emaneeventservice just to add the extra functionality.

I know ihmc have a java event emitter, so I might beg permission to access its source code to see if it's more easily modifiable.