adjacentlink/emane #211: freespace propagationmodel doesn't work for EMANE 1.2.7

martin20082018, 2022-03-24:
In this little example - from github with small modifications using sdt3d and change the position to lat/lon/alt
The code works when the "2ray" propagation model is used.
If I change to "freespace" propagtion model (see line with # <<<<<<<<<) NO LINK is established anymore between the 2 nodes.
I use EMANE 1.2.7 and I alreday check new newest code on github - the same code.

Can you PLEASE double check this on your side?
Maybe I also do something wrong...

THANK YOU FOR YOUR EFFORTS!

```
# required imports
from core.emane.ieee80211abg import EmaneIeee80211abgModel
from core.emane.nodes import EmaneNet
from core.emulator.coreemu import CoreEmu
from core.emulator.data import IpPrefixes, NodeOptions
from core.emulator.enumerations import EventTypes
from core.nodes.base import CoreNode
import subprocess
import logging

logging.basicConfig(level=logging.DEBUG)

# ip nerator for example
ip_prefixes = IpPrefixes(ip4_prefix="10.0.0.0/24")

# create emulator instance for creating sessions and utility methods
#coreemu = CoreEmu()
coreemu = CoreEmu({"controlnet": "176.16.0.0/24"})
session = coreemu.create_session()
session.sdt.initialize()

# location information is required to be set for emane
session.location.setrefgeo(53.960087, 9.617760, 600)
session.location.refscale = 150.0

# conect to SDT3D
session.sdt.connect()

# must be in configuration state for nodes to start, when using "node_add" below
session.set_state(EventTypes.CONFIGURATION_STATE)

# create emane
options = NodeOptions(emane=EmaneIeee80211abgModel.name)
options.set_location(53.960087, 8.617760, 600)
emane = session.add_node(EmaneNet, options=options)

# create nodes
options = NodeOptions(model="mdr")
options.set_location(53.370592, 7.772324, 250)
n1 = session.add_node(CoreNode, options=options)
session.services.add_services(n1, n1.type, ["SSH"])

options = NodeOptions(model="mdr")
options.set_location(53.363261, 6.806689, 400)
n2 = session.add_node(CoreNode, options=options)
session.services.add_services(n2, n2.type, ["SSH"])

# configure general emane settings
config = session.emane.get_configs()
config.update({"eventservicettl": "2"})

# configure emane model settings
# using a dict mapping currently support values as strings
session.emane.set_model_config(
    emane.id, EmaneIeee80211abgModel.name, {
        "unicastrate": "3",
#        "propagationmodel": "freespace"                             # <<<<<<<<<<<<<<<<<<<<<<<<
    }
)

# link nodes to emane
iface1 = ip_prefixes.create_iface(n1)
session.add_link(n1.id, emane.id, iface1)
iface1 = ip_prefixes.create_iface(n2)
session.add_link(n2.id, emane.id, iface1)

# start session
session.instantiate()
subprocess.check_call("sdtcmd layer " + "CORE::Links,on",shell=True)
# do whatever you like here
input("press enter to shutdown")

# stop session
session.shutdown()

```

sgalgano, 2022-03-24:
2-ray pathloss works out to be 92.32 dB and freespace (assuming 2GHz) is 134.63 dB. At that point it is a matter of transmit power and antenna gains to determine receiver power and then SINR, and a curve lookup for probability of reception.

martin20082018, 2022-03-24:
Thank you very much! My fault! 
I double checked it with "freespace" and "precomputed" - for the later I used the python code from your Demonstration1 in the emane tutorial - all of them worked!

closed 2022-03-24 by martin20082018