adjacentlink/emane #216: EMANE Pre-built Distribution Package for RHEL 8 lxc.init.static error

roymatchuu, 2022-05-05:
I am trying to build EMANE v1.3.3 on a vanilla RHEL 8 machine and have ran into a few troubles. I have been following the steps detailed [here](https://github.com/adjacentlink/emane/wiki/Install#rhelrocky-8-x86_64) to build on a vanilla RHEL 8 machine. 

I am able to follow the steps until I get to the `dnf config-manager --set-enabled powertools` step where I have used [this resource](https://access.redhat.com/discussions/5417621) to solve the problem by performing a `sudo subscription-manager repos --enable codeready-builder-for-rhel-8-x86_64-rpms` instead. 

I also encounter problems on the `dnf install *.rpm` where I am not able to install python3-pynodestatviz due to an incorrect dependency name (expected as: python3-pmw). I am able to build pynodestatviz [from source](https://github.com/adjacentlink/pynodestatviz) and install pmw by performing a `python3 -m pip install pmw` call. 

I then clone emane-tutorial and try to run demo 0 to verify that everything works. I then install `bridge-utils` to get the olsrlinvkview.py to work, but the demo logs show a `No such file or directory - unable to open lxc.init.static` error in the `lxc-execute.log.1`

When looking up this problem, I found this bug report: https://bugzilla.redhat.com/show_bug.cgi?id=1654366

The bug report then prompted me to start from scratch and perform `yum install glibc libcap` before running the `dnf install *.rpm` step. The same lxc.init.static error persists even after rebuilding with the supposed libraries needed to configure `HAVE_STATIC_LIBCAP` to yes. What steps should I take to solve this problem in a Vanilla RHEL 8 system?

sgalgano, 2022-05-05:
This is an lxc rpm spec issue. Download the lxc source rpm, modify the spec similar to below, rebuild, and update.

```diff
--- lxc.spec.orig	2022-05-05 16:58:12.932812621 -0400
+++ lxc.spec.new	2022-05-05 16:44:50.353538858 -0400
@@ -1,3 +1,5 @@
+%define buildid .static_init
+
 %if 0%{?fedora}
 %global with_seccomp 1
 %global with_static_init 1
@@ -6,12 +8,13 @@
 %if 0%{?rhel} >= 7
 %ifarch %{ix86} x86_64 %{arm} aarch64
 %global with_seccomp 1
+%global with_static_init 1
 %endif
 %endif
 
 Name:           lxc
 Version:        3.0.4
-Release:        2%{?dist}
+Release:        2%{?buildid}%{?dist}
 Summary:        Linux Resource Containers
 License:        LGPLv2+ and GPLv2
 URL:            http://linuxcontainers.org/lxc
```

roymatchuu, 2022-05-05:
Can you point me towards where I can find the spec file you have specified I need to make changes into? I can't seem to match the spec file sample you have there with [this spec file](https://github.com/lxc/lxc/blob/lxc-3.0.4/lxc.spec.in) from v3.0.4 tag of LXC.

roymatchuu, 2022-05-05:
I was able to find the source RPM for lxc v3.0.4 from here: https://koji.fedoraproject.org/koji/buildinfo?buildID=1374546

I edited the SPEC file to reflect the changes you had specified but I am now having trouble rebuilding due to the following errors:
```
error: Failed build dependencies:
	libcap-static is needed by lxc-3.0.4-2.static_init.el8.x86_64
	pkgconfig(libseccomp) is needed by lxc-3.0.4-2.static_init.el8.x86_64
```

I cannot seem to find any reference online on how to install libcap-static especially for the RHEL 8 system. I am also having trouble finding the libseccomp.pc that is expected by the SPEC file in the /usr/share/pkgconfig folder after installation. Do you have any references on where I can find these dependencies?

sgalgano, 2022-05-06:
You can download source rpms with dnf:

```
$ dnf download --source lxc
```

The dependencies were a bit more difficult to find -- we rebuilt them from fedora src rpms. 

Pre-built *lxc-3.0.4-2.static_init* package bundles for convenience [[Package Signature](#package-signature)]:

| Distro        | Bundle                                        | sha256sum |
|---------------|-----------------------------------------------|-----------|
|RHEL/Rocky 8|[64-bit](https://adjacentlink.com/downloads/emane/lxc-3.0.4-2.static_init-release-1.el8.tar.bz2) [*(sig)*](https://adjacentlink.com/downloads/emane/lxc-3.0.4-2.static_init-release-1.el8.tar.bz2.asc)|0ae0cc7c858f49dede10ad22ab0f4d577ab6651b082383368fd7ad1a93c855c4|

roymatchuu, 2022-05-09:
Would it be proper to download the Pre-built lxc-3.0.4-2.static_init package you had attached and install all the RPMs in the RPMs directory? After having installed everything:

- lxc-templates-3.0.4-2.static_init.el8.x86_64.rpm
- lxc-libs-3.0.4-2.static_init.el8.x86_64.rpm
- lxc-devel-3.0.4-2.static_init.el8.x86_64.rpm
- lxc-3.0.4-2.static_init.el8.x86_64.rpm

The `No such file or directory - unable to open lxc.init.static` error is gone but the `Function not implemented - Failed to send signal through pidfd` still persists.

sgalgano, 2022-05-09:
Yes.

```
$ sudo rpm -i lxc-libs-3.0.4-2.static_init.el8.x86_64.rpm lxc-3.0.4-2.static_init.el8.x86_64.rpm
```

Keep in mind the version of olsr used in the [emane-turtorial](https://github.com/adjacentlink/emane-tutorial) is no longer maintained, so mileage will vary.

roymatchuu, 2022-05-09:
I still get the `Function not implemented - Failed to send signal through pidfd` after performing the command you had mentioned above. Even though the `unable to open lxc.init.static` error is resolved, the `Function not implemented - Failed to send signal through pidfd` still persists.