adjacentlink/emane #222: Beginner's Questions

fxie00, 2022-10-04:
Hi,

I am new to EMANE. I have some basic questions with respect to adjusting delay and packet loss in EMANE. I start with RF pipe model. Thanks in advance for any help.

1. I used emaneshell to adjust the MAC layer delay. However, it is NOT the expected propagation delay. It is accumulated and added to every packet (e.g. the delay for the first N packets will be t, 2t, 3t, …. Nt). This causes timeout for end-to-end ping/tcp sessions. What I need is a propagation delay, which is equal to every packet over the emulated wireless channel. Further, I want to adjust such a delay on the fly while EMANE is running. Are there any ways to do this?

2. EMANE uses pcr table to determine the packet loss. Can I change the table (sinr/por mapping) on the fly while EMANE is running? If not, are there ways that I can affect the SINR calculation on the fly, such as adjust transmitter power? I tried to change the txpower via emaneshell but the change seemed to have no impact on the packet loss.

Thanks again for the help.

Feng

kalhansb, 2024-10-22:
> Hi,
>
> I am new to EMANE. I have some basic questions with respect to adjusting delay and packet loss in EMANE. I start with RF pipe model. Thanks in advance for any help.
>
> 1. I used emaneshell to [...]

I have the same question. Help would be much appreciated!

sgalgano, 2024-10-22:
What version of emane are you using?

kalhansb, 2024-11-13:
Sorry for the delay!
I was considering using EMANE with Core emulator.
My question exact question is: 
Can we adjust receiving power for a node using some other script?
I am looking to integrate Core + EMANE with Gazebo.