adjacentlink/emane #231: Clarification on EMANE tables

vivianchiong, 2023-03-08:
What is the difference between tables with the same name, like "BroadcastPacketAcceptTable0" in IEEE 802.11abg Model and the PHY layer model? I noticed they can give different results when using emanesh.

Also, what do the values of 0, 1, 2 3, tacked onto these tables mean like in "BroadcastPacketAcceptTable0"?

sgalgano, 2023-03-09:
Drop and accept tables provide layer specific information such as traffic counts (bytes and packets) that have successfully traversed through the layer (accept tables) or counts broken out by reason for why traffic was dropped (drop tables). Drop reasons are different for each layer.

The index at the end of the table name corresponds to queue index.

vivianchiong, 2023-03-10:
Makes sense, thank you so much!

closed 2023-03-10 by vivianchiong