adjacentlink/emane #243: [Question] Different PCR curve, data rate, and packet size for separate nodes?

James-Sloop, 2024-02-23:
I am interested in developing my own EMANE model and so far I have been working with the 802.11 and RFPipe models to learn EMANE. RFPipe has a single PCR curve, packet size, and data rate used across the emulation and it seems like the 802.11 has multiple PCR curves, a single packet size, and a single data rate. The 802.11 uses just 1 PCR curve at a time (the same for every node) and doesn't change the data rate index dynamically based on network performance. Is this understanding correct? I am curious if it is possible to have different nodes within the emulation use different PCR curves and data rates or if those have to be emulation-wide parameters. Is there an easy way to configure as a model plug-in or will it require modifying EMANE? If it does require modifying EMANE, how difficult would that be to do and could anyone point me in the right direction to get that started? Thank you for the help.

patelkb, 2024-02-25:
EMANE framework does not need to be modified to support what you want to 
do.  Your radio model implementation would contain everything you 
mention you want to do.

It is up to the radio model implementation to determine how it handles 
receive packet processing as it pertains to packet completion rates, 
dyanmic rate adaptation and any other other features the waveform is 
required to emulate.

If the model you are developing supports N different modulations and 
each modulation has varying bit error rate curves, your model should 
have a separate curve to capture that behavior.  In theory, you are not 
required to use curves.  It is up to you to implement that processing 
however you want to achieve it.

As far as dynamic rate/modulation adaptation goes, none of the open 
source models has dynamic rate adaptation.   For the 802.11 model, each 
node can be configured to use a different rate as long as they are all 
configured to use the same mode.  The model also allows you to change 
that rate runtime.  The feature to dynamically change the rate based on 
network conditions is not implemented within the model.

So if your model requires dynamic rate adaptation, you would implement 
that within your model using your specific algorithm. In the case of the 
802.11 model, you can actually have a separate application outside of 
the model that gets access to transmit and receive metrics via the EMANE 
control port and dynamically determines what rate each node should be 
using and update the configuration via the control port.

kb

Kaushik B. Patel
Adjacent Link LLC
CEO, Director of Engineering

James-Sloop, 2024-02-26:
This was a very helpful response, thank you so much.

closed 2024-02-26 by James-Sloop