adjacentlink/emane #244: Problem with the emanesh command

James-Sloop, 2024-02-28:
I am trying to work with EMANE and would like to be able to modify the unicastrate and multicastrate of the IEEE802.11abg model at runtime. When I tried to use ```$ emanesh node-1 set config nems mac unicastrate=10``` I get an error. I get the same error when I try to run ```emanesh node-1``` without any other parameters.
```
jcsloop in ~:emanesh node-1
Traceback (most recent call last):
  File "/usr/bin/emanesh", line 4, in <module>
    __import__('pkg_resources').run_script('emane==1.5.1', 'emanesh')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3267, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3241, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3279, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 573, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 891, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 777, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'emane==1.5.1' distribution was not found and is required by the application
jcsloop in ~:
```
I am certain my emane version is 1.5.1:
```
jcsloop in ~:emane --version
1.5.1
jcsloop in ~:
```

I am not sure what could be causing this problem. I have tried to solve the problem myself by making sure emanesh has been installed correctly:
```
jcsloop in ~:which emanesh
/usr/bin/emanesh
jcsloop in ~:
```

I have also confirmed emanesh is in my $PATH. I have tried adding the path to emanesh into my $PYTHONPATH and that didn't solve the issue either. Both of my coworkers working with me are having the same problem where we are just unable to use emanesh which has made working with EMANE at runtime quite problematic. Other than emanesh not working everything else has been working normally so I don't think it is a problem with my EMANE installation. My coworkers and I are all using Ubuntu 22.04 with EMANE version 1.5.1. Is there a known fix for this issue?

sgalgano, 2024-02-28:
How did you install emane?

sgalgano, 2024-02-28:
Your issue is more than likely related to the way you installed emane. CORE is a 3rd party tool that we do not use or support. My suggestion would be to install emane from debs on Ubuntu 22.04.

bharnden, 2024-02-28:
> I cloned it from `https://github.com/adjacentlink/emane.git` when I was installing CORE. I used these installation instructions:
> 
> ```
> # install emane
> cd ~/Documents
> wget https://github.com/protocolbuffers/protobuf/releases/download/v3.19.6/protoc-3.19.6-linux-x86_64.zip
> mkdir protoc
> unzip protoc-3.19.6-linux-x86_64.zip -d protoc
> git clone https://github.com/adjacentlink/emane.git
> cd emane
> ./autogen.sh
> ./configure --prefix=/usr
> make -j$(nproc)
> sudo make install
> cd src/python
> make clean
> PATH=~/Documents/protoc/bin:$PATH make
> sudo /opt/core/venv/bin/python -m pip install .
> ```
> 
> Everything but `emanesh` is currently working fine and I talked to the people at CORE and they said this is an EMANE issue unrelated to CORE.

I did not say this was an EMANE issue, but as noted above. An issue with how you have installed EMANE currently.

James-Sloop, 2024-02-29:
Ok I will try installing EMANE a different way and see if that fixes the issue.